Barracuda09/PyPSADiag #20: Ulp file s3 line checksum calculate

tontito, 2024-12-25:
hi how to calculate ulp file s3 line cheksum

    def calcCRC16X25(self, data: str):
        if len(data) & 1:
            return ["BE", "EF"]
        byteData = []
        for i in range(0, len(data), 2):
            byteData.append(int(data[i:i + 2], 16))
        crc = 0xFFFF
        for i in range(0, len(byteData)):
            crc ^= (byteData[i] & 0xFF) << 0
            for _ in range(8):
                if crc & 0x0001:
                    crc = (crc >> 1) ^ 0x8408
                else:
                    crc = crc >> 1
        print(f'CRC: {crc}')
        crc = crc ^ 0xFFFF
        return ["%0.2X" % (crc & 0xFF), "%0.2X" % ((crc >> 8) & 0xFF)]

S38580080200BB34D3BF0000000065A9C6AC50CEF3CBA3AB00AE52EFF1EA5543F646FA1A591F060BA50E205C835988432B465D16FE13CE7A6D7F2F1D8C18E2B241B7C5F266F7E4914794754AD64F92428942077D83EDE27899175363B79C326C460CA7644C940212161DFB1C70836206E999C70ED301B20559F5133E675E823F66C067201C4F6E

calculate this line 

S3 85 80080200 BB34 D3BF0000000065A9C6AC50CEF3CBA3AB00AE52EFF1EA5543F646FA1A591F060BA50E205C835988432B465D16FE13CE7A6D7F2F1D8C18E2B241B7C5F266F7E4914794754AD64F92428942077D83EDE27899175363B79C326C460CA7644C940212161DFB1C70836206E999C70ED301B20559F5133E675E823F66C06720 1C4F 6E
this line crc16/x-25 checksum 1C4F but this code result : 0CB2

how to calculate ulp s3 line checksum calculate?

16D, 2024-12-25:
This checksum is added for coding DAE_BVH2 in program, so it should work

16D, 2024-12-26:
This line is from can sniffer during teletransmission?

Barracuda09, 2024-12-26:
This online calculator comes to the same result: `B20C` [Calc](https://crccalc.com/?crc=D3BF0000000065A9C6AC50CEF3CBA3AB00AE52EFF1EA5543F646FA1A591F060BA50E205C835988432B465D16FE13CE7A6D7F2F1D8C18E2B241B7C5F266F7E4914794754AD64F92428942077D83EDE27899175363B79C326C460CA7644C940212161DFB1C70836206E999C70ED301B20559F5133E675E823F66C06720&method=CRC-16/X-25&datatype=hex&outtype=hex), Probably you need to add more data for the checksum?

So that makes me believe that the Python CRC16/X25 calculation is correct.

tontito, 2024-12-27:
İşte eklediğiniz kısmı da içeren tam İngilizce çeviri:

Hello, I am uploading the PSA ECU ULP update file to Google Drive. In the Cal calibration files, the checksum result at the end of short lines is correct, but in the ULP files, the checksum we calculate does not match the one at the end of the lines in any case. Online CRC calculations and your code are correct, but the checksum at the end of the lines is different. I have included the files as examples.

This file is the original update file for the Citroen C5 Aircross ECU, obtained via the official diagnostic tool, Diagbox.

https://drive.google.com/file/d/1CSryg8Z0Jj6J5ZgZtjOOY_kogBWT8YQA/view?usp=drive_link

Barracuda09, 2024-12-28:
It almost makes me think, there is no CRC16/X25 in the .ulp file at all. And that it should maybe calculated at the moment it is send?

Did you retrieve the file from the temp directory that Diagbox uses?

tontito, 2024-12-28:
Yes, I copied it from the temp folder while updating diagbox online.

16D, 2024-12-28:
I should try record teletransmission of some module but not with vlud adapter (don't record commands send to module). I think you prepare wrong line to send. Read this carefully:
https://github.com/ludwig-v/arduino-psa-diag/blob/master/UDS_FLASH.md

16D, 2024-12-28:
Here is teletransmission sniffed and used file:
https://drive.google.com/file/d/1lqIojDtwjQUDC9VtnpGIggbAxyoh-yGf/view?usp=drive_link
[teletrans.txt](https://github.com/user-attachments/files/18266556/teletrans.txt)

tontito, 2024-12-29:
What I want to do is to make changes in the ulp file and turn off immo or egr. Here is the sample file, I can find the checksum2 of the immo line, but I cannot match the checksum part with your codes, CRC-16/X-25

https://drive.google.com/file/d/1fa6_TikgxGA56FY_h1UER_kkvuSBfjQx/view?usp=drive_link

16D, 2024-12-29:
I think this is other checksum like in this link:
https://github.com/ludwig-v/arduino-psa-diag/tree/master?tab=readme-ov-file#content-data

tontito, 2024-12-29:
> I think this is other checksum like in this link: https://github.com/ludwig-v/arduino-psa-diag/tree/master?tab=readme-ov-file#content-data

I already did this by trying. The short one is s2, but it does not work on long lines or s3 lines.

Barracuda09, 2024-12-30:
How do you know what to change in the ulp file to disable that? by Disassemble?
Does this file not not download or is there some other problem or you are trying to make a download program?

yoursbofh, 2025-07-07:
> How do you know what to change in the ulp file to disable that? by Disassemble? Does this file not not download or is there some other problem or you are trying to make a download program?
> 
> Sorry, I do not know what you are trying todo here

I'm trying to check if the ulp/cal file is correct before start unit flashing , according to     https://github.com/ludwig-v/arduino-psa-diag/tree/master?tab=readme-ov-file#content-data 

CHECKSUM calculated as CRC-16/X-25(DATA) with this order CRC[1] CRC[0] but using this algo this doesn't match.
for example line from (correct) ulp file:
S20F000202646E82969EA09B1432**3C4B**5C - should be 
S20F000202646E82969EA09B1432**5872**5C - calculated (should be not 5C at the end but it doens't matter in case of checksum calculation).

I think that's exactly what the author meant (algorithm for S3 lines the same)

Barracuda09, 2025-07-07:
The CRC16/X25 is only used when sending to ECU.

The checksum for "inside" ulp files (SREC) is different see:
https://en.m.wikipedia.org/wiki/SREC_(file_format)