Barracuda09/PyPSADiag #35: ECU unlock: Failed, ECU Reports Invalid Key (7F2735) ECU unlock: Failed, ECU Reports Invalid Key (7F2735)

lepromer, 2025-04-13:
hello 
firstly your share is very best, thank you, i hope you will more improve your project,
but i have a problem , why dont work in telecodng, i am getting this error:
 ECU unlock: Failed, ECU Reports Invalid Key (7F2735)

16D, 2025-04-13:
What ECU you tried to unlock?

lepromer, 2025-04-13:
İ tried coding ESP90 for automatic parking

16D, 2025-04-13:
You tried all 3 keys?

lepromer, 2025-04-13:
Did I not notice the 3 keys?, where are they, how can i try thwy

lepromer, 2025-04-13:
Can You share any photo where are they

Barracuda09, 2025-04-13:
[image]

lepromer, 2025-04-13:
I really appreciate, i am trying just now

lepromer, 2025-04-13:
İ tried all keys but it is giving still error

[image]

Barracuda09, 2025-04-13:
OK, then probably the ESP90 you have uses a different key.

Do you know for sure it is a ESP90?

lepromer, 2025-04-13:
Probably,my vehicle DS7 2019 year and 1. 5 Diesel 130hp
Thank you your helps

Barracuda09, 2025-04-15:
Are you sure it is a ESP90?

Could you paste or add CSV file of your ECU reading?

kt819a, 2025-04-22:
> İ tried coding ESP90 for automatic parking

Try key ABFB for ESP81(by Ludwig)

lepromer, 2025-04-22:
i will try it, when i wnet to anothet city, i couldn't yet

Barracuda09, 2025-04-24:
From which car is this?

lepromer, 2025-04-24:
Vehicle,Ds7, i tried ESP90 keys, but i got error again, later i decided changing in the Json code the key ABFB and i didnt get error, in addition my car's ECU different i deced rhe key 1948 and, it telecoded, but i dont know how can i activated semi automatic park

Barracuda09, 2025-04-25:
You mean for Engine ECU (MD1CS003) you do not know how to set it up for semi-parking assist?

Sorry, at the moment I have no info on this ECU (INJ)

lepromer, 2025-04-25:
Vehicle DS7 1.5 130hp eat-8, it has cvm3 cam.
Engine ECU (MD1CS003)
2100 0547FA6600FEFF88442BC403
2101 05EFFB7FBFFFFFFE44EBFD
ESP
2100 06000202B4FB210000000040C0000000A420200B8CE3FFE0A01004FF01
2101 06FBFFFFFFFFFFFFFDFFFFFFFFEFFFFFFFFFF77FFFEFFFFFEFBA7EFF

thank you MARC

ygorigor, 2025-05-14:
> Vehicle DS7 1.5 130hp eat-8, it has cvm3 cam. Engine ECU (MD1CS003) 2100 0547FA6600FEFF88442BC403 2101 05EFFB7FBFFFFFFE44EBFD ESP 2100 06000202B4FB210000000040C0000000A420200B8CE3FFE0A01004FF01 2101 06FBFFFFFFFFFFFFFDFFFFFFFFEFFFFFFFFFF77FFFEFFFFFEFBA7EFF
> 
> thank you MARC

@lepromer do you still need help to activate parking assist? 
I can try to help you with that.

lepromer, 2025-05-14:
> [@lepromer](https://github.com/lepromer) do you still need help to activate parking assist? I can try to help you with that.

Yes please, thank You for your help

ygorigor, 2025-05-14:
> Yes please, thank You for your help

You have to activate parking assistance options in NAC, BSI and AAS.

Disel11, 2025-05-14:
> > > > Автомобиль DS7 1.5 130 л.с. eat-8, имеет cvm3 cam. ЭБУ двигателя (MD1CS003) 2100 0547FA6600FEFF88442BC403 2101 05EFFB7FBFFFFFFFE44EBFD ESP 2100 06000202B4FB210000000040C0000000A420200B8CE3FFE0A01004FF01 2101 06FBFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF77FFFEFFFFFFFEBA7EFF 
> > > > спасибо MARC
> > > 
> > > 
> > > [@lepromer](https://github.com/lepromer) тебе все еще нужна помощь с активацией парковочного ассистента? Я могу попытаться помочь тебе с этим.
> > 
> > 
> > Да, пожалуйста, спасибо за помощь.
> 
> Вам необходимо активировать опции помощи при парковке в NAC, BSI и AAS.
> 
> В BSI, Тип помощи при парковке, я выбрал опцию 73, помощь при парковке спереди и сзади + камера помощи при движении задним ходом + полуавтоматическая система парковки с возможностью перпендикулярной парковки.

bonjour. pouvez-vous m'aider aussi? j'ai remplacé le propulseur d'étrave avec un capteur d'angle de rotation des roues (le mien ne l'était pas). je veux activer LKA. inclus la présence du capteur dans BSI, DAE_BVH2, NAC, CIROCCO. mais dans Diagbox ne voit pas le capteur (erreur pas de communication). avez-vous une idée de ce qu'il faut faire?  si vous avez des idées, pouvez-vous laisser votre E-mail pour la communication?

Disel11, 2025-05-14:
> Вам необходимо активировать опции помощи при парковке в NAC, BSI и AAS.
>
> В BSI, Тип помощи при парковке, я выбрал опцию 73, помощь при парковке спереди и сзади + камера помощи при движении задним [...]

здравствуйте, может у вас получится и мне помочь? я заменил свой подрулевой блок (в старом не было датчика угла поворота колес). хочу активировать подруливание LKA. но diagbox пишет, что нет связи с датчиком. откалибровать тоже не дает. в BSI, DAE_BVH2, NAC, CIROCCO вроде все включено. предохранитель F2 стоит. что может быть не так? может оставите свою почту для общения?

> > > > Vehicle DS7 1.5 130hp eat-8, it has cvm3 cam. Engine ECU (MD1CS003) 2100 0547FA6600FEFF88442BC403 2101 05EFFB7FBFFFFFFE44EBFD ESP 2100 06000202B4FB210000000040C0000000A420200B8CE3FFE0A01004FF01 2101 06FBFFFFFFFFFFFFFDFFFFFFFFEFFFFFFFFFF77FFFEFFFFFEFBA7EFF
> > > > thank you MARC
> > > 
> > > 
> > > [@lepromer](https://github.com/lepromer) do you still need help to activate parking assist? I can try to help you with that.
> > 
> > 
> > Yes please, thank You for your help
> 
> You have to activate parking assistance options in NAC, BSI and AAS.
> 
> In BSI, Type of parking assistance, I chose the option 73, front and rear parking assistance + reversing assistance camera + semi-automatic parking system with perpendicular parking option

здравствуйте, может у вас получится и мне помочь? я заменил свой подрулевой блок (в старом не было датчика угла поворота колес). хочу активировать подруливание LKA. но diagbox пишет, что нет связи с датчиком. откалибровать тоже не дает. в BSI, DAE_BVH2, NAC, CIROCCO вроде все включено. предохранитель F2 стоит. что может быть не так? может оставите свою почту для общения?

ygorigor, 2025-05-15:
> здравствуйте, может у вас получится и мне помочь? я заменил свой подрулевой блок (в старом не было датчика угла поворота колес). хочу активировать подруливание LKA. но diagbox пишет, что нет связи с [...]

Можно в телеграм @ygorigor

lepromer, 2025-05-15:
> You have to activate parking assistance options in NAC, BSI and AAS.
> 
> In BSI, Type of parking assistance, I chose the option 73, front and rear parking assistance + reversing assistance camera + semi-automatic parking system with perpendicular parking option

@ygorigor thank you , I will try it at the first opportunity..

pe308, 2025-05-16:
hello friends
dae-bvh2 dynamic drive block is missing
has an edit been made regarding this?

Barış DURUKAN

SAYGILARIMLA
BEST REGARDS

lepromer ***@***.***>, 15 May 2025 Per, 11:19 tarihinde şunu
yazdı:

z3r0trac3, 2025-05-26:
Hello m i have also a ds7 and i want to enable ambient lighting m but i get error when trying to telecode   , anyone can help

2025-05-26 16:12:47.069965 --|  < 500100C80014
2025-05-26 16:12:47.078065 --|  Reading ECU Zones: Successful
2025-05-26 16:12:55.645635 --|  > >752:652
2025-05-26 16:12:55.783733 --|  < OK
2025-05-26 16:12:56.293752 --|  > KU
2025-05-26 16:12:56.408386 --|  < OK
2025-05-26 16:12:56.416231 --|  > 1003
2025-05-26 16:12:56.533332 --|  < 500300C80014
2025-05-26 16:12:57.049622 --|  > 2703
2025-05-26 16:12:57.174338 --|  < 7F2722
2025-05-26 16:12:57.184158 --|  ECU Unlock Request: Failed (7F2722)
2025-05-26 16:12:57.193780 --|  > S
2025-05-26 16:12:57.312680 --|  < OK
2025-05-26 16:12:57.319465 --|  > 1001
2025-05-26 16:12:57.439466 --|  < 500100C80014
2025-05-26 16:13:07.644293 --|  > >752:652
2025-05-26 16:13:07.782835 --|  < OK
2025-05-26 16:13:08.311637 --|  > KU
2025-05-26 16:13:08.431858 --|  < OK
2025-05-26 16:13:08.443271 --|  > 1003
2025-05-26 16:13:08.564800 --|  < 500300C80014
2025-05-26 16:13:09.091459 --|  > 2703
2025-05-26 16:13:09.202380 --|  < 7F2722
2025-05-26 16:13:09.214347 --|  ECU Unlock Request: Failed (7F2722)
2025-05-26 16:13:09.226714 --|  > S
2025-05-26 16:13:09.344958 --|  < OK
2025-05-26 16:13:09.356404 --|  > 1001
2025-05-26 16:13:09.474826 --|  < 500100C80014
2025-05-26 16:13:17.173630 --|  > >752:652
2025-05-26 16:13:17.323325 --|  < OK
2025-05-26 16:13:17.844853 --|  > KU
2025-05-26 16:13:17.953096 --|  < OK
2025-05-26 16:13:17.965660 --|  > 1003
2025-05-26 16:13:18.080125 --|  < 500300C80014
2025-05-26 16:13:18.600420 --|  > 2703
2025-05-26 16:13:18.606404 --|  < 7F3E03  ** Skipping **
2025-05-26 16:13:18.832184 --|  < 7F2722
2025-05-26 16:13:18.844185 --|  ECU Unlock Request: Failed (7F2722)
2025-05-26 16:13:18.856431 --|  > S
2025-05-26 16:13:18.985723 --|  < OK
2025-05-26 16:13:18.997385 --|  > 1001
2025-05-26 16:13:19.119786 --|  < 500100C80014

PTZ-M, 2026-10-05:
I’m changing 2 parameters in the BCM — verins_motorises — zone 2100 from a Peugeot 3008, 2018 model year. When writing, I get a similar error.

z3r0trac3, 2026-10-06:
> I'm changing 2 parameters in the BCM - verins_motorises - zone 2100 from a Peugeot 3008, 2018 model year. When writing, I get a similar error.

Engine is started , or in economy mode

16D, 2026-10-06:
> I’m changing 2 parameters in the BCM — verins_motorises — zone 2100 from a Peugeot 3008, 2018 model year. When writing, I get a similar error.

for this module key is D702, edit json