Barracuda09/PyPSADiag #36: Se puede virginizar una bsi2010ev

arobles91, 2025-04-23:
Hola, me pueden indicar si puedo virginizar una bsi2010 ve con este programa, muchísimas gracias de antemano.

Barracuda09, 2025-04-23:
No, solo puedes cambiar las zonas (todas a la vez si quieres). Pero no "reiniciar" el kilometraje.

English:
No you can only change the zones (all at once if you like).
But not "reset" mileage.

arobles91, 2025-04-23:
no quiero reiniciar el kilometraje quiero cambiar el pin code ya que lo puse mal

Disel11, 2025-05-14:
hello. I have a problem reading the BSI block. reading stops almost immediately and after that he doesn't even want to try to read the zones. reads other blocks. everything is working fine on the other computer. what kind of problem could be on my computer?

[image]