Barracuda09/PyPSADiag #46: Witness echo in cirocco

farias1830, 2025-07-30:
It would be interesting to implement the echo button configuration in CIROCCO, which in some boxes is located in the last bits of zone 2101. This encoding is not found in the VLUD application nor in Alicar ACT V2 and some users have to code it manually. Example of zone 2101 with echo tester activated on pins 13 and 14 of this CIROCCO. 2101 = 0000 0000 0000 0601 0404 0106 0000 00 02 0707 00 0505

Barracuda09, 2025-07-30:
Do you have more info on zone 2101 and the rest of the bytes?
Do you know how to change the JSON files (CIROCCO.json)?

farias1830, 2025-07-30:
This is what I can provide for manually editing zone 2101 in CIROCCO.
I'm not familiar with JSON files.
I can also confirm that not all CIROCCOS have those last bits in zone 2101 to encode the indicator and echo button. I imagine this will depend on the calibration installed, but I still need to confirm that.

[metodo edicion campo 2101 sirocco2.pdf](https://github.com/user-attachments/files/21514265/metodo.edicion.campo.2101.sirocco2.pdf)