Barracuda09/PyPSADiag #47: Timeout when connecting

kapollo, 2025-08-05:
I'm constantly getting Timouts when trying to connect with your tool. I'm choosing for instance BSI, choose proper key, and connect to the arduino and I'm getting:

[image]
I'm using it on a bench, vlud's script is connecting just fine if I do it by hand. Is this something known? I'm using Python 3.13.3

Barracuda09, 2025-08-05:
Which vlud ino did you use?
Did you connect to the COM port?

kapollo, 2025-08-05:
Hi I've used the main script the 500k one, I've connected to the BSI with console using this script just fine it responds and so on. 
And yes I was using the correct COM port.
That's what is making me curious... I will try to setup everything on another computer, this one might have some "baggage" although I've used virtualenv...
I will report if it works or not, by the way do you have some verbose debug switch I could turn on, perhaps this might help a bit (I'm not really a python coder, but I can read it well enough).
Thanks for you interest btw

kapollo, 2025-09-01:
Sorry I got distracted with life..
I still have the issue, I have no clue what is happening, but there is a workaround for this. Before I click on the Connect I enter R into the send box, right after I click connect I click on the send button (second button click get's dequeued as soon control returns to the gui), and then it catches the OK from the arduino and works.
But it's strange as hell, it happens on both bench, and the "regular" script. At the begging I thought it's a problem of encoding or new line char but, hell I don't know right now. I will try to investigate it further in spare time.
I originally asked the question because I thought it was a common enough issue :)
Btw. great work and great help.
Cheers !